UMSAE-Formula-Electric/SCU #22: Implement SD Card

Sedkian, 2023-07-26:
This task includes:

- writing the necessary code for the implementation of the SD Card into the SCU.
- investigating what is wrong with the SD Card.
- writing the logging functions and getting them ready to be used in other C files.

Sedkian, 2023-08-16:
https://embetronicx.com/tutorials/microcontrollers/stm32/stm32-sd-card-interfacing-with-example/

moram03, 2024-02-14:
Currently the SD card is having issues mounting. Some reasons for this may be as follows:

- Issue with devboard -> Highly unlikely as I have tried using other boards and the issue persisted
- Issue with the SD card -> definately not as it works fine with my computer and other microcontroller I own
- Issue with the SD card reader -> possible, I have been trying to use it with other microcontrollers, however have not found a straightforward guide to getting SD cards working using SDIO on other boards
- Issue with the software library -> I really hope not, this will be a pain to isolate and fix
- Issue with our code -> Probably not as code taken from tutorials is basically the same (we have different print statements and the like), but it is not impossible

NikoChristie, 2024-02-16:
https://01001000.xyz/2020-08-09-Tutorial-STM32CubeIDE-SD-card/
https://learn.adafruit.com/adafruit-micro-sd-breakout-board-card-tutorial/arduino-wiring
https://os.mbed.com/platforms/ST-Nucleo-F446RE/